MikePopoloski/slang #897: Enhancing VCS Streaming Operator Compatibility

shoulihs, 2024-02-21:
Hi there,

First off, thanks so much for the incredible work on Slang! I'm truly impressed by how quickly the parser operates and its minimal memory usage. I'm eager to integrate Slang into my project, but I've encountered an issue.

**Is your feature request related to a problem? Please describe.**

I've come across an error message stating:
```
streaming operator can only be used in an assignment or bit-stream cast argument
```
Here's the original SV code I'm using:
```verilog
let known(sig) = (!($isunknown({>>{sig}})));

property known_prop(sig, clk);
  @(clk) (!($isunknown({>>{sig}})));
endproperty
```
Upon reviewing the LRM, it seems this usage should be invalid unless the signal type is a bit-stream type. Nevertheless, VCS allows it, albeit with a warning:
```
Warning-[ETRFSFAC] Explicit typecast required for streams for assertion context
```
This approach is prevalent in our codebase, making it challenging to modify the existing code.

**Describe the solution you'd like**

Given that Slang has a `--compat vcs` flag for compatibility with VCS, it would be greatly appreciated if the tool could allow this expression when VCS compatibility mode is enabled.

Thanks for considering this request!

MikePopoloski, 2024-02-22:
Hmm, I guess the VCS behavior here is to compute a total width of the stream elements as a self-determined expression and then pretend there is a cast to a logic vector of that same width? It seems doable. Curious why you're using this construct as opposed to just using the signal directly? This doesn't actual change any of the bits right?

shoulihs, 2024-02-23:
I didn't code this, but I believe the intent was to create a universal function for both unpacked and packed arrays, to be used in an assertion library. `$isunknown` can only take a packed array, so it uses the streaming operator to convert an unpacked array into a stream, which becomes a packed array.

MikePopoloski, 2024-03-08:
Added a compatibility flag in a34e29b4810ea5f8982b5111452be4826d8cc65d

closed 2024-03-08 by MikePopoloski