MikePopoloski/slang #900: Adds a possibility to access the first Token of a SyntaxNode.

suzizecat, 2024-02-23:
### Discussed in https://github.com/MikePopoloski/slang/discussions/898

As discussed, there would be a need to write-access in a generic fashion the first token of any `SyntaxNode`.

I am willing to try and implement it myself.

In order to not disrupt the current features would be to:
 - Create a `ReferenceTokenOrSyntax` object (or another name) which would hold the reference or a pointer to a `Token` or a `Syntax`.
 - Update `gen_syntax.py` to implement the `getChild` version which would return the `ReferenceTokenOrSyntax`
 - Implement/Update the `getFirstChild` and `getLastChild` functions. 

Could I have a confirmation that this is the right approach ?

MikePopoloski, 2024-02-25:
Yes, that sounds right. I think you'll want it to be a pointer, so TokenPtrOrSyntax probably.

suzizecat, 2024-02-25:
Ok, thanks for the pointers.
I may not have much time in the next week, but I'll have a look into this 🙂

closed 2024-03-01 by MikePopoloski