MikePopoloski/slang #901: Expected Integer Base Apecifier After Signed Specifier

shoulihs, 2024-02-28:
**Describe the bug**
A simple example that triggers an error is provided. The 'signed specifier' should not be checked within a string.

**To Reproduce**
```systemverilog
`define xcheck_msg `"There are X's in signal`"

module Test;
    initial begin
        $display(`xcheck_msg);
    end
endmodule
```

output:
```
test.sv:3:35: error: expected integer base specifier after signed specifier
`define xcheck_msg `"There are X's in signal`"
                                  ^
```

yanggeorge, 2024-02-29:
I find another question. 
```
`define H(x) "Hello, x"
`H(world)
```
After running `slang --preprocess`, here is output
```
"Hello, x"
```
IMO, ``` `H(world) ``` should be expanded to `Hello, world`  from LRM 22.5.1, not including double quotes.

yanggeorge, 2024-02-29:
@MikePopoloski  I was wrong.

closed 2024-02-29 by MikePopoloski