MikePopoloski/slang #906: Model resolution in libraries

Krym4s, 2024-03-06:
If several files have modules with similar names, slang cannot distinguish it and inline it in elaborated model. This problem is related to 33.6 chapter of LRM.

Example:
```
// file top.v
config cfg1;
design top;
default liblist rtlLib, aLib, gateLib;
endconfig

module top();
logic a;
adder a1(a);
adder a2();
endmodule

//file adder.v

module adder(logic a);
...
endmodule;

//file adder.vg

module adder(logic a);
...
endmodule;

//file lib.map
library rtlLib top.v
library aLib adder.v
library gateLib adder.vg
```
 
```
slang top.v adder.v adder.vg --libmap=lib.map
```

Module <em>top</em> in code above should find module adder in library aLib and different module adder in gateLib. Slang finds both module definition, but have problem with building elaborated model and throws an error.

MikePopoloski, 2024-03-06:
Are you using master or the 5.0 release? Support for configurations was added after 5.0 was released.

Krym4s, 2024-03-06:
Hello, we run this example on the master branch.

Krym4s, 2024-03-06:
Code example was slightly changed.

MikePopoloski, 2024-03-06:
You need to tell slang to use the configuration, so something like `slang top.v --libmap=lib.map --top cfg`

Krym4s, 2024-03-06:
Thank you, everything is working now.