MikePopoloski/slang #910: Can randomize()/std::randomize have getEffectiveWidth return 1?

jrudess, 2024-03-09:
This is a continuation of info from #909 

I believe that randomize()/std::randomize() fall into a similar bucket where the LRM defines the return value as int type, but the only possible return values are 0 or 1.

This causes code like the following to flag truncation warnings.  Can 'getEffectiveWidth' be setup for this build-in to return 1 similar to the exists() built in for associative-arrays?

```
slangtest157.sv:4:16: warning: implicit conversion truncates from 32 to 1 bits [-Wwidth-trunc]
        bit rc = std::randomize(x);
               ^ ~~~~~~~~~~~~~~~~~.
```
```systemverilog
class C;
    function bit f();
        bit x;
        bit rc = std::randomize(x);
        assert (rc);
        return x;
    endfunction
endclass
```

jrudess, 2024-03-10:
$cast() also falls into this bucket of returning 0 or 1 as int type.

MikePopoloski, 2024-03-10:
$cast was already handled but I tagged some other ones in 8336fc639ecf8c5eb0cd27b57b3293511e99947e

closed 2024-03-10 by MikePopoloski