MikePopoloski/slang #911: generate-else after assert property is terminated with semi-colon is linked to that assert-property causing parse failure

jrudess, 2024-03-09:
This is a tricky one that took me a while to understand to be able to reproduce.

For the generate-if that wraps label1, I believe the generate-else is being incorrectly associated with the prior assert-property.  The semi-colon at the end of that line should have completely terminated the 'assert property' so the else should not be associated with it.   The examples for label3->label6 showcase some workarounds to get the parsing to work out.

```
slangtest158.sv:16:9: error: cannot have 'assert' statement in the action block of another concurrent assertion
        label2: assert property(myprop(k));
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
```systemverilog
module M #(
    A = 1
);
logic clk, rst;

property myprop(k);
   @(posedge clk) disable iff(rst !== 0) k > 0;
endproperty

genvar k;
for (k=0; k < 4; k++) begin: m
    if (A)
        label1: assert property(myprop(k));
    else
        label2: assert property(myprop(k));

    if (A)
        label3: assert property(myprop(k))
                else $error("assert failed");
    else
        label4: assert property(myprop(k));

    if (A) begin
        label5: assert property(myprop(k));
    end else
        label6: assert property(myprop(k));
end

endmodule
```

jrudess, 2024-03-09:
As a side-note, I've seen cases where VCS accepts 'else $error()' statements appearing after an assert-property is terminated with a semicolon, but then Questa fails to parse.  It looks like Slang might be matching VCS behavior here as the following example parses cleanly.

```systemverilog
module M;

logic clk, rst;

property myprop;
   @(posedge clk) disable iff(rst !== 0) 1;
endproperty

label1: assert property(myprop);
        else $error("with semi-colon");

label2: assert property(myprop)
        else $error("without semi-colon");

endmodule
```

MikePopoloski, 2024-03-10:
Fixed in 1ed2b7a93ce88da803f4611dda11731da8a73bf1

closed 2024-03-10 by MikePopoloski