MikePopoloski/slang #922: slang macro

hongping, 2024-03-28:
**Describe the bug**
slang cannot handle when the argument for macro has '*'

**To Reproduce**
Source code:
```
`define import_base(_pkg)\
    import _pkg;

module test7();
    `import_base(uvm_pkg::*)
endmodule
```

slang will error as:
tests/unittests/data/test7.sv:4:16: error: expected 'endmodule'
module test7();
               ^
tests/unittests/data/test7.sv:4:1: note: to match this 'module'
module test7();
^
tests/unittests/data/test7.sv:6:10: error: expected ')'

hongping, 2024-03-28:
the problem is whenever the source code has (* or *), it is lex as OpenParenthesisStar or StarCloseParenthesis, which makes the macro expansion can't find ( or ).

MikePopoloski, 2024-03-29:
Fixed in 9705113788c5a11749ae463f3ee3e49751948666

closed 2024-03-29 by MikePopoloski