MikePopoloski/slang #924: Non-standard ANSI port definition

adream307, 2024-03-30:
**Is your feature request related to a problem? Please describe.**
According to the SystemVerilog standard, this code should  report error, but both VCS and Cadence support this syntax.
```sv
module top(input a, output b);
wire [3:0] a;
reg [3:0] b;
endmodule
```

**Describe the solution you'd like**
Under the `--compat vcs` option, this syntax is supported, otherwise an error will occur.

This feature has been widely used in previous code, so I really look forward to suporting it, thanks.

MikePopoloski, 2024-04-06:
Hmm, I could add this but even VCS 2021.09 says this:
```
Warning-[IPDASP] Identifier in ANSI port declaration
testbench.sv, 4
  Redeclaration of ANSI ports not allowed for 'a', this will be an error in a 
  future release
```
So it seems you're going to need to fix your code at some point anyway when VCS gets around to making this an error. Is that not possible? Are you planning on staying on an old VCS version forever?

adream307, 2024-04-09:
Thanks for your reply.

> Hmm, I could add this but even VCS 2021.09 says this:
>
> ```
> Warning-[IPDASP] Identifier in ANSI port declaration
> testbench.sv, 4
>   Redeclaration of ANSI ports not allowed for 'a', this will be an [...]

adream307, 2024-04-09:
> Hmm, I could add this but even VCS 2021.09 says this:
>
> ```
> Warning-[IPDASP] Identifier in ANSI port declaration
> testbench.sv, 4
>   Redeclaration of ANSI ports not allowed for 'a', this will be an [...]

Thanks for your reply.
Yes, you are right ,we should not staying on an old VCS version foever.
I'm  using slang as synthesis tool's parser, and this feature exists widely in existing synthesizable  code, DC also support this feature.
This exists code is provided as `IP`, It may not run on VCS ,but it would be compiled by DC. 
So I think my customers would begin to fix their code if DC also report error on this feature.
It would be really helpful if slang support this feature, thanks.

MikePopoloski, 2024-04-09:
Ok, I am not against adding a compatibility flag for this, but it's also not at the top of my priority list at the moment.

MikePopoloski, 2024-05-29:
Added in 895805e5bceaf9d405eb6769710529f0fba8e2bf

closed 2024-05-29 by MikePopoloski