MikePopoloski/slang #927: slang allows Verilog code to redefine `define's from the command line

udif, 2024-04-01:
**Describe the bug**
While not part of the spec, Most of the tools (or at least the 2 that I have checked), treat preprocessor defines that were defined by the command line args (-define or `+define+<NAME>=<val>`) to be "stronger" than those defined within Verilog files.

**To Reproduce**
```Verilog
`define N 1
module t;
generate
  $info("%d", `N);
endgenerate
endmodule
```
Call this with:

`slang +define+N=2 t.v`

And you would get:
```
Top level design units:
    t

t2.v:4:3: note: $info encountered:           1
  $info("%d", `N);
  ^

Build succeeded: 0 errors, 0 warnings
```

Run the equivalent code on one tool and you would get:
```
...
	Elaborating the design hierarchy:
	Top level design units:
		t
          2 
  $info("%d", `N);
...
```
Another tool yields:
```
# KERNEL: Info: testbench.sv (6):           2
```

**Additional context**
You may consider this behavior to be optional (using a new flag), if you think changing this would break past usage of slang, in the same way you do with `--single-unit`, which as I have just noticed,  conforms the IEEE spec to the letter (3.12.1):

_The exact mechanism for defining which files constitute a compilation unit is tool-specific. However, compliant tools shall provide use models that allow **both** of the following cases:_
_a) All files on a given compilation command line make a single compilation unit (in which case the declarations within those files are accessible following normal visibility rules throughout the entire set of files)._
_b) Each file is a separate compilation unit (in which case the declarations in each compilation-unit scope are accessible only within its 
 corresponding file)._

udif, 2024-04-02:
Two more data points (from my tests with these 2 simulators):
1. If your code undefines the macro in your verilog file, it **will** get undefined, regardless of its origin (command line or within your code).
2. If you redefine it again after undefining it, the new value overrides the command line value (e.g. once it is undefined, it is really gone).

MikePopoloski, 2024-04-02:
All four tools I tried also exhibit the same behavior, so it makes sense for slang to just have the same behavior as well (not gated behind a flag).

closed 2024-04-02 by MikePopoloski