MikePopoloski/slang #930: Non reproducible output?

udif, 2024-04-02:
**Describe the bug**
slang output is not reproducible on every run.

**To Reproduce**
I can't reproduce this here, because the output is the result of an inhouse design with 1000+ files, but before I investigate further, I want to understand first if reproducible output for the same input is guaranteed at all.

**Additional context**
The design is very large, but here are the key issues:  
There are still dozens of errors, but what I see is that the run always starts by reporting some errors from 2 files which are always the same.
After that, it may report or skip errors from 2 other files. It then keep reporting some more errors from other files, where it stops after 20 errors (at places, due to errors skipped in one run, having a lower error count).

The errors which are skipped or not skipped are caused by usage of packages:
Two modules, `a` and `b`, each defines a package, `pkg_a` and `pkg_b` respectively.
both packages define a localparam with the same name, `PARAM`.
Both modules import their respective package outside the module definition.  
The modules then use `PARAM` without qualifying it as `pkg_a::PARAM` (or b) , at which point slang errors on the 2nd module because PARAM is already defined (we use `--single-unit` semantics).
```Verilog
package pkg_a;
localparam PARAM=16;
...
```

```Verilog
package pkg_b;
localparam PARAM=16;
...
```

```Verilog
import pkg_a::*;
module a;
...
wire [PARAM-1:0]x;
```

```Verilog
import pkg_b::*;
module b;
...
wire [PARAM-1:0]x;
```

I'm not asking whether this is legal or not, but I'm wondering why this is inconsistent. Sometimes I get this error, sometimes not.

**UPDATE:**
I just ran this with `--errors-limit=100` and it seems like a file order issue.
Could it be that slang doesn't always reads the files in the same order?
For the `--single-unit` mode, file order is critical.

MikePopoloski, 2024-04-02:
In theory builds should be reproducible when run with --single-unit. slang will try to perform parsing in parallel where possible but it's not actually possible to do that when --single-unit is in use. Even when it does do parallel parsing, each parsed unit is independent from the others and all diagnostics emitted are collected and sorted before issuing, so again in theory it should always result in the same output. That doesn't mean I don't have a bug somewhere though.

You can test whether threads might be the problem here by using `-j 1` to disable any multithreading logic.

udif, 2024-04-02:
Well, if you depend on a macro defined by one file to be available in the next one, you can't do parallel processing.
I haven't noticed slang supports `-j` (nice!). That definitely explains everything.
What are the boundaries for parallel processing? At the file level?

MikePopoloski, 2024-04-02:
As I mentioned, if you're using --single-unit then the files aren't parsed in parallel, so I'm not sure it explains anything. Otherwise, parsing is parallelized at the level of an individual compilation unit, which by default is each source file but can be customized using compilation unit files, library files, etc. Also the actual file loading (the reading of bytes from disk into memory) will be done in parallel but that shouldn't affect output.

udif, 2024-04-02:
I can confirm that `-j 1` Makes the result fully reproducible.
The main issue is defines, which affect the code even at the module scope.
The only way I can think of to solve such a thing is to have one thread quickly do preprocessing, and then do parallel parsing.

MikePopoloski, 2024-04-02:
Preprocessing is part of "parsing", for purposes of these discussions. When using --single-unit, each source file is pushed onto a single Preprocessor class's stack and then lexed / preprocessed / parsed as if it were a single large file concatenated together.

Since `-j 1` does fix the problem for you it seems I have a bug somewhere, likely in SourceLoader.cpp.

udif, 2024-04-02:
I think I know what the issue is.

I have a "vendor compatibility layer" `.f` file that includes a bunch of flags (mostly `--cmd-ignore` and `--cmd-rename`)  to filter our whole tree of nested `.f` files, so that slang won't choke on vendor specific flags.
This file allows me to take our top level `.f` file as-is without preprocessing and run it against slang.

One of the first flags in the file is `--single-unit`. I don't speficy it explicitly on my command line.

Front what you describe, this would arrive too late for slang to switch to single parsing thread.
I'll try adding `--single-unit` on the command line instead of `-j 1` and see if it has the same effect.

MikePopoloski, 2024-04-02:
No, all command line args, including stuff from .f files, is rolled up together before trying to load or parse anything.

Do you pass files using any other mechanism besides just listing them? Like specifying them as library files using -v or using compilation unit listings with -C or library maps or whatever?

udif, 2024-04-02:
What we usually do is to have a tree of `-f` files, containing:
1. Various tool flags
2. `+incdir` definitions
3. `+define` macro definitions
4. Verilog/SystemVerilog files

All of these are mixed in no particular order as needed.

Verilog/SystemVerilog files fall into 1 of 5 types:
1. Regular modules - 1 per file, with module name matching filename.
2. Files only containing macro defines - these are listed as part of the `.f` files before they are used.
3. Files containing function definitions. These are included by other Verilog files using the `include` preprocessor directive, because they need to be within a module scope.
4. Vendor IP - we just make sure we have the correct `+incdir` is set as required, and let the vendor do what they prefer. (No rules).
6. packages, one package per file, with filename matching package name - these are included as part of the `-f` filelist.

We don't use any library mechanism, or SystemVerilog configuration/Libmap files.

MikePopoloski, 2024-04-06:
When you say that the output is non-deterministic, do you mean that sometimes the build passes and sometimes it does not, or do you mean that you get the same set of diagnostic output each time, just sometimes in a different order (note that you might need --error-limit=0 to see all of it).

If it's the latter, then I know what's happening and it's sort of expected; diagnostics are sorted by their location, and locations sort based on the ID of the buffer they point to, and the buffer ID is essentially an arbitrary counter assigned in whatever order the file is loaded. So in that sense the build itself is fully reproducible but the order of the output is not. I can fix that pretty easily though.

MikePopoloski, 2024-04-06:
d9ac28a2bc0a28f805b75c7b6c02e1e28d1a385c should fix the diagnostic sorting

udif, 2024-04-07:
From the limited tests I've done, this seems to solve the problem.
All error logs are now reproducible without resorting to `--j 1`.
Thanks!

closed 2024-04-07 by MikePopoloski