MikePopoloski/slang #949: Add caching of identical instance bodies

MikePopoloski, 2024-04-07:
Support caching instance bodies and skipping over identical copies when elaborating, to save time. Expose this functionality so that downstream consumers of the AST can easily do the same.

Note that the tricky detail here is figuring out how to handle hierarchical references into and out of the cached instances.

yanggeorge, 2024-04-29:
It is better that elabarated AST only show one instance of identical instances.

likeamahoney, 2024-09-16:
My hash suggestion - https://github.com/MikePopoloski/slang/pull/1088#discussion_r1761208918

MikePopoloski, 2025-01-04:
This can now be turned on with a compilation flag, with a caveat that it can cause some multi-driven errors to not be issued correctly. Things to handle before it's fully robust:
- Assignments through interface ports need to be reapplied to specific instances being connected
- Any upward hierarchical references that extend out of the cached instance need to be re-resolved, and the resolved types must match, otherwise we can't use the cache entry
- Downward hierarchical references into such instances need to be accounted for
- Bind directives needs to be accounted for
- Defparams need to be accounted for
- Configuration rules need to be accounted for

MikePopoloski, 2025-02-17:
This is complete.

closed 2025-02-17 by MikePopoloski