MikePopoloski/slang #956: Initial value of automatic function return value

udif, 2024-04-08:
**Describe the bug**
Not sure if this is a bug, and what to do about it.

**To Reproduce**
```
module t;
  function automatic [31:0] test();
    for (int i = 0; i < 3; i++) begin
      test = test + i;
    end
  endfunction
  
  generate
    $info("%d", test());
  endgenerate
endmodule

```
**Additional context**
slang returns X for the code above.
2 other commercial tools are split between x and 3.
I haven't checked yet what the IEEE1800-2023 LRM says.
For my code I have simply initialized test to 0, but it was caught only by slang, while our simulator happily accepted it with no error.

Can we issue an explicit warning if a constant function is evaluating a local variable without initializing it first?

udif, 2024-04-11:
Try this with all 4, you will be surprised (set SV mode ofcourse)...
```
module t;
  function automatic [31:0] test();
    for (int i = 0; i < 3; i++) begin
      test = test + i;
    end
  endfunction
  
  generate
    $info("%d", test());
  endgenerate

  initial
    $info("%d", test());

endmodule
```

MikePopoloski, 2024-04-11:
I tried it; RivieraPro, Xcelium, Questa, and VCS all print x twice. Not sure what you were expecting to happen?

udif, 2024-04-11:
I don't like mentioning vendor names, but ...

Look at the output from the `$info()` call, yielding `3` instead of `x` (I don't know if I can mark something within a code section in MD).

`slang` actually caught this while `Xcelium` silently generated good code for the wrong reasons.
You have to remove all the compiled files because it happens only inside `generate` sections at compile time.

Anyway, from what you describe, this is definitely an Xcelium bug, given that this behavior is seen by a single simulator at a specific corner case, and it seems to contradict what we expect according to thew standard.

```bash
% cat t.v
module t;
  function automatic [31:0] test();
    for (int i = 0; i < 3; i++) begin
      test = test + i;
    end
  endfunction
  
  generate
    $info("%d", test());
  endgenerate

  initial
    $info("%d", test());

endmodule

% rm -rf xcelium.d/
% xrun -sv t.v
TOOL:	xrun	23.03-s005: Started on Apr 11, 2024 at XX:XX:XX XXX
xrun: 23.03-s005: (c) Copyright 1995-2023 Cadence Design Systems, Inc.
file: t.v
	module worklib.t:v
		errors: 0, warnings: 0
		Caching library 'worklib' ....... Done
	Elaborating the design hierarchy:
	Top level design units:
		t
         3 
    $info("%d", test());
        |
xmelab: *N,INFOIE (./t.v,9|8): Elaboration encountered a $info at t.
	Building instance overlay tables: .................... Done
	Generating native compiled code:
		worklib.t:v <0x4b68a494>
			streams:   2, words:   858
	Building instance specific data structures.
	Loading native compiled code:     .................... Done
	Design hierarchy summary:
		         Instances  Unique
		Modules:         1       1
		Registers:       4       2
		Initial blocks:  1       1
	Writing initial simulation snapshot: worklib.t:v
Loading snapshot worklib.t:v .................... Done
xcelium> source /apps/tools/Cadence/XCELIUMMAIN/23.03.005/tools/xcelium/files/xmsimrc
xcelium> run
xmsim: *N,INFSEV (./t.v,13): (time 0 FS).
t
         x
xmsim: *W,RNQUIE: Simulation is complete.
xcelium> exit
TOOL:	xrun	23.03-s005: Exiting on Apr 11, 2024 at XX:XX:XX XXX  (total: 00:00:00)
```

MikePopoloski, 2024-04-18:
Going to close this as it's not a slang bug. The data flow analysis work doesn't need to be tracked by this issue, it's a much broader project.

closed 2024-04-18 by MikePopoloski