MikePopoloski/slang #967: tools/netlist unittest failure for v6.0 package release

jrudess, 2024-04-22:
The packaging process for Arch runs the full unit tests, and while prepping the new 6.0 tag for a package update, the netlist DepthFirstSearchTests has a failure:

```
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
netlist_unittests is a Catch2 v3.5.3 host application.
Run with -? for options

-------------------------------------------------------------------------------
Depth-first search with an edge predicate
-------------------------------------------------------------------------------
/home/jrudess/repos/slang-verilog/src/slang-6.0/tools/netlist/tests/DepthFirstSearchTests.cpp:94
...............................................................................

/home/jrudess/repos/slang-verilog/src/slang-6.0/tools/netlist/tests/DepthFirstSearchTests.cpp:111: FAILED:
  CHECK( *visitor.nodes[1] == n2 )
with expansion:
  {  } == {  }

/home/jrudess/repos/slang-verilog/src/slang-6.0/tools/netlist/tests/DepthFirstSearchTests.cpp:112: FAILED:
  CHECK( *visitor.nodes[2] == n4 )
with expansion:
  {  } == {  }

===============================================================================
test cases:  41 |  40 passed | 1 failed
assertions: 280 | 278 passed | 2 failed
```

MikePopoloski, 2024-04-22:
Hmm weird, the CI builds run all the tests via ctest as well. Somehow this is specific to Arch? Or it's non-deterministic behavior in the test that just so happens to work on all the CI runners.

jrudess, 2024-04-22:
Just noticed that 'unittests' is failing also with a timeout at 60s.  I'll debug the failures more this week.

gcc = 13.2.1
cmake = 3.29.2

```
cmake -DCMAKE_CXX_COMPILER=g++ -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr -DBUILD_SHARED_LIBS=True -DCPACK_PACKAGE_NAME=svlang -DCPACK_PACKAGE_INSTALL_DIRECTORY=svlang -DCMAKE_PROJECT_NAME=svlang ..
```

yanggeorge, 2024-04-23:
I have tried gcc-11-release and macos-release presets, and all tests of unittests and netlist_unittests passed.

jrudess, 2024-04-24:
I can clone, build, run all tests fine using when done manually.  The problem only starts once the build process is wrapped inside the Arch packaging flow.  I will track down (eventually) what is causing a difference between build environments.   If I can concretely identify a compile or linker switch that reproduces this behavior, then will reopen.

closed 2024-04-24 by jrudess