MikePopoloski/slang #968: The bit/part-select results should be unsigned. But not in slang v3.0.

hailongSun2000, 2024-04-22:
Excuse me, sir. When I tested the vector bit-select and part-select operations in slang v3.0, I found their results were not as described.
Example:  (Please see IEEE 1800-2017 § 11.8.1 "Rules for expression types".)
## .sv
```
module Foo;
  logic [15:0] a;
  logic signed [7:0] b;
  initial
          a = b[7:0];
endmodule
```
## .json
```json
{
  "name": "$root",
  "kind": "Root",
  "addr": 2199024449984,
  "members": [
    {
      "name": "",
      "kind": "CompilationUnit",
      "addr": 2199025895968
    },
    {
      "name": "Foo",
      "kind": "Instance",
      "addr": 2199025898560,
      "body": {
        "name": "Foo",
        "kind": "InstanceBody",
        "addr": 2199025897488,
        "members": [
          {
            "name": "a",
            "kind": "Variable",
            "addr": 2199025897680,
            "type": "logic[15:0]",
            "lifetime": "Static"
          },
          {
            "name": "b",
            "kind": "Variable",
            "addr": 2199025898056,
            "type": "logic signed[7:0]",
            "lifetime": "Static"
          },
          {
            "name": "",
            "kind": "ProceduralBlock",
            "addr": 2199025898432,
            "procedureKind": "Initial",
            "body": {
              "kind": "ExpressionStatement",
              "expr": {
                "kind": "Assignment",
                "type": "logic[15:0]",
                "left": {
                  "kind": "NamedValue",
                  "type": "logic[15:0]",
                  "symbol": "2199025897680 a"
                },
                "right": {
                  "kind": "Conversion",
                  "type": "logic[15:0]",
                  "operand": {
                    "kind": "Conversion",
                    "type": "logic signed[15:0]",
                    "operand": {
                      "kind": "RangeSelect",
                      "type": "logic signed[7:0]",
                      "selectionKind": "Simple",
                      "value": {
                        "kind": "NamedValue",
                        "type": "logic signed[7:0]",
                        "symbol": "2199025898056 b"
                      },
                      "left": {
                        "kind": "IntegerLiteral",
                        "type": "int",
                        "value": "7",
                        "constant": "7"
                      },
                      "right": {
                        "kind": "IntegerLiteral",
                        "type": "int",
                        "value": "0",
                        "constant": "0"
                      }
                    }
                  }
                },
                "isNonBlocking": false
              }
            }
          }
        ],
        "definition": "Foo"
      },
      "connections": [
      ]
    }
  ]
}
```

MikePopoloski, 2024-04-24:
Agreed, this looks like a bug.

MikePopoloski, 2024-04-27:
Fixed in 14ccafa6edb721cc41b08179c486975ecffb7a74

closed 2024-04-27 by MikePopoloski

hailongSun2000, 2024-04-28:
> Fixed in [14ccafa](https://github.com/MikePopoloski/slang/commit/14ccafa6edb721cc41b08179c486975ecffb7a74)

Thanks so much for fixing this bug!