MikePopoloski/slang #980: Array with size from body parameter is <error>

pehnetom, 2024-05-03:
**Describe the bug**
When a array size is defined by body parameter, then elaborated symbol will have type `<error>`.

**To Reproduce**
Following SV source code results in the bug:
```
interface test_interface();
    parameter int C_PARAM_PARAM = 42;
    logic [1:C_PARAM_PARAM] value_1;
    logic value_2;
    logic value_3;
    integer integer_4;
endinterface : test_interface
```

**Additional context**
This maybe happens as a value of the parameter symbol is `<unset>` and the actual value is kept inside initializer.
Problem detected with using pyslang==5.0.0 and pyslang==6.0

MikePopoloski, 2024-05-05:
It sounds like you're inspecting an uninstantiated instance of the interface? slang will automatically create these to ensure that the body gets some checking even if the interface is unused in the design, but parameters are forced to invalid values since it's not a real instantiation and no param values were provided. You can see this in the API by looking at the `isUninstantiated` member of the instance body.

If you inspect an actual instance of that interface it should have the correct types.

MikePopoloski, 2024-05-09:
Going to close this since there's been no response. Feel free to reopen if you have further questions.

pehnetom, 2024-05-10:
Sorry for the late repsonse.

What you wrote is correct. I created a dummy top module to wrap uninstantiated modules/interface, which solved the problem for me. Thanks!

shivang-chipstack, 2024-07-01:
> It sounds like you're inspecting an uninstantiated instance of the interface? slang will automatically create these to ensure that the body gets some checking even if the interface is unused in the [...]

Thank you for the explanation on parameterized interfaces. However, rather than assuming some invalid/forced value for an uninstantiated interface, we should provide with default value in type. To give little more context, consider below code:

```
interface my_intf # (
  IN_WIDTH = 8,
  OUT_WIDTH = 8
  )
  ();
  logic [IN_WIDTH-1:0] d_in;
  logic [OUT_WIDTH-1:0] d_out;
  
  modport sp (input d_in, output d_out);
  modport mp (output d_in, input d_out);
endinterface

module DUT(
	input logic clk,
	input logic reset,
	my_intf.sp intf_inst
);
	logic [7:0] d_flop1;
	always @(posedge clk) begin
		if (reset) begin
			intf_inst.d_out <= 0;
			d_flop1 <= 0;
		end else begin
			d_flop1 <= intf_inst.d_in;
			intf_inst.d_out <= d_flop1;
		end
	end

endmodule
```

For above example, when we check the type of my_intf.sp.d_in, it returns <error>. Technically, we should return <logic [7:0]> as we can assume default value of IN_WIDTH=8 from interface definition. If the actual DUT instance is provided then we can refer to instantiated interface width, but when there is no instantiation provide we should use default value rather than <error> for interface signal type.

Can you please update and add this support?