MikePopoloski/slang #982: --ast-json-depth <x> flag

AndrewNolte, 2024-05-04:
Only print the json for a scope up to this module depth. I'm requesting this because the json files can quickly get way too huge to manage.

Also, should the ast printer maybe be split out into a different tool?

MikePopoloski, 2024-05-05:
I'm not against adding this. In general though the JSON output is not likely to ever be usable for real production scale projects precisely for this reason. Ideally slang would gain an alternative format (or more than one) that is more compact and useful for real world designs.

The serialization printing could live in another tool but I'm not sure you gain much by doing that. It's also nice if the `slang` tool can emit things instead of just printing diagnostics.

AndrewNolte, 2024-05-08:
I was also thinking about adding more info emitters for a design, like ast, filelist, used includes, potential defines (or a .f file with all three), instance counts, etc. I think if more of these were added it would pollute the flag namespace for the linter.

I think it would make sense to have these live in a tool called slang-query, which can be merged with slang-hier and slang-netlist. I'd also like to start on slang-ls, and it would make sense for this to share components of slang-query.

MikePopoloski, 2024-05-09:
That sounds fine to me. I was thinking that it would be nice to have something like iverlog's -M flag to print dependencies to a file. Having a more comprehensive tool for inspection would be good too.