MikePopoloski/slang #1006: Integrate to Neovim as a language server

MinhTran0911, 2024-05-23:
I am wondering if this can be used as a language server for Neovim via Neovim’s LSP to provide linting/code completion/syntax highlighting? If it is possible then how to set it up?

udif, 2024-05-23:
Have you tried:
https://danielmangum.com/posts/setup-verible-verilog-neovim/
?

I think [Verible ](https://github.com/chipsalliance/verible)is more suitable for this task:
> The Verible project's main mission is to parse SystemVerilog (IEEE 1800-2017) (as standardized in the [SV-LRM](https://ieeexplore.ieee.org/document/8299595)) for a wide variety of applications, including developer tools.

> It was born out of a need to parse _un-preprocessed_ source files, which is suitable for single-file applications like style-linting and formatting. In doing so, it can be adapted to parse preprocessed source files, which is what real compilers and toolchains require.

MikePopoloski, 2024-05-24:
slang works fine as the engine for a language server; some things I know about via a quick search:
- https://alvinrolling.github.io/vim/Slang-Linter-for-Vim-Ale/
- https://github.com/vivekmalneedi/veridian
Hopefully that gets you started.

closed 2024-05-24 by MikePopoloski

sequencer, 2024-09-07:
Maybe this one is better to use pure slang as language server?
https://github.com/suzizecat/slang-lsp-tools
veridian uses verible, which doesn't work well with complex Verilog designs.