MikePopoloski/slang #1017: [bug][slang] Incorrect empty ports.

Kitaev2003, 2024-05-31:
**Describe the bug**
Hello, is this code correct? If so, maybe you should add diagnostics for this code or mark these ports as invalid?

**To Reproduce**
```verilog

module top( , , );

endmodule
```

```json
{
"name": "top",
"kind": "Instance",
"addr": 2199025898640,
"body": {
    "name": "top",
    "kind": "InstanceBody",
    "addr": 2199025898352,
    "members": [
    {
        "name": "",
        "kind": "Port",
        "addr": 2199025898784,
        "type": "void",
        "direction": "In"
    },
    {
        "name": "",
        "kind": "Port",
        "addr": 2199025898960,
        "type": "void",
        "direction": "In"
    },
    {
        "name": "",
        "kind": "Port",
        "addr": 2199025899136,
        "type": "void",
        "direction": "In"
    }
    ],
    "definition": "2199025942528 top"
},
"connections": [
]
}
```

MikePopoloski, 2024-06-02:
Yes, the code is valid. There is an `isNullPort` member that is not included in the JSON serialization but easily could be.

MikePopoloski, 2024-06-05:
Added in ad5cda219803a4829c70a2e4bea24d04591a32c3

closed 2024-06-05 by MikePopoloski