MikePopoloski/slang #1021: RHS width reported incorrectly for -Wsign-conversion

jrudess, 2024-06-07:
In the following example, the warnings are reporting a type as being bit[31:0], but the variable is bit[10:0].

```
slangtest166.sv:12:15: warning: implicit conversion changes signedness from 'bit[31:0]' to 'int' [-Wsign-conversion]
        s.a = e;
            ~ ^
slangtest166.sv:15:16: warning: implicit conversion changes signedness from 'bit[31:0]' to 'int' [-Wsign-conversion]
            a: g ? 10'h0 : e
            ~  ^~~~~~~~~~~~~
slangtest166.sv:19:16: warning: implicit conversion changes signedness from 'bit[31:0]' to 'int' [-Wsign-conversion]
            a: e
            ~  ^
```
```Systemverilog
class C;
    typedef struct {
        int a;
    } s_t;

    function f();
        s_t s;
        bit [10:0] e;
        bit g;

        s.a = 0;
        s.a = e;

        s = '{
            a: g ? 10'h0 : e
        };

        s = '{
            a: e
        };

        s = '{
            a: 0
        };

    endfunction
endclass
```

MikePopoloski, 2024-06-08:
The warnings are on the post-expansion type. Since the lhs is 32 bits wide, the rhs implicitly converts to a 32 bit size along with the sign conversion. I suppose you could make the argument that either way is more understandable, depending on how you think about how conversions work, but the current situation is not obviously wrong to me.

jrudess, 2024-06-09:
That makes sense, thanks!

closed 2024-06-09 by jrudess