MikePopoloski/slang #1022: False degenerate error when using within-statement

jrudess, 2024-06-07:
I believe the below example is a case of a false degenerate report, or if it were degenerate and could never emit a match, I would expect simulation failures for this assertion with VCS.
```
slangtest167.sv:9:22: error: any sequence that is used as a property shall be nondegenerate and shall not admit any empty match
        $fell(a) |-> !$stable(b) within((~a)[*1:$] ##1 a);
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
slangtest167.sv:7:14: note: while expanding property 'p1'
    property p1(clk, rstb, a, b);
             ^
slangtest167.sv:9:22: note: sequence property can never be matched
        $fell(a) |-> !$stable(b) within((~a)[*1:$] ##1 a);
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
```Systemverilog
module top;
    logic clk;
    logic rstb;
    logic a;
    logic b;

    property p1(clk, rstb, a, b);
        @(posedge clk) disable iff(~rstb)
        $fell(a) |-> !$stable(b) within((~a)[*1:$] ##1 a);
    endproperty

    LABEL1: assert property(p1(clk, rstb, a, b));
endmodule
```

likeamahoney, 2024-06-07:
Hi! Thanks for the bug report!

I opened the PR which fixes that - https://github.com/MikePopoloski/slang/pull/1023

MikePopoloski, 2024-06-09:
Fixed in 2122d639817cca5234d377baa2e1b2aa8abf4592 instead

closed 2024-06-09 by MikePopoloski