MikePopoloski/slang #1032: Support for legacy protection pre-processor directives

rahultadak, 2024-06-15:
Based on issue #612  and my experience with slang v6.0, it doesn't seem to support older code protection directives described below:
```
`protect
...
`endprotect
```
. After encrypting source code tagged as above it is turned into:
```
`protected
... <base64 blob>...
`endprotected
```

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

MikePopoloski, 2024-06-15:
Is there documentation anywhere about how these directives work exactly? They aren't covered in the LRM -- are they literally identical to the corresponding pragma directives?

rahultadak, 2024-06-16:
That is a good question. I've not actually created protected code using this, but I've seen protected code failing compile with slang. I could probably recreate this and put in a log here.

In the meantime, this is the closest I've found to a description on how this might work.

https://redirect.cs.umbc.edu/portal/help/VHDL/verilog/compiler_directives.html

rahultadak, 2024-06-18:
Without going into too much detail, I have something like this in a file:
```
`protected
...
...
`endprotected
```

Trying to run it with slang like this:
```
$ slang file.svp

file.svp:line:1: error: unknown macro or compiler directive '`protected'
`protected
^
```

I removed the details of the files and line numbers.

sequencer, 2024-06-18:
We also encountered this under the SNPS VIP flow.

MikePopoloski, 2024-06-22:
Added in 2daa108eede929b7e5fabe8b2e09402898627f04

closed 2024-06-22 by MikePopoloski

udif, 2024-07-11:
Just wanted to thank you for this change. I just joined a new project still using IP with this old protection method, and it was so nice being able to update slang and have it parse the file flawlessly.