MikePopoloski/slang #1044: crashes when using slang as a library.

automatic-robot, 2024-07-02:
Hello Mike:
I'm hoping you can shed some quick insight into a crash I'm seeing when I use the slang_package_example with the slang library
being installed somewhere on the filesystem.  I was trying to adapt your slang-tidy application with a separate build flow because I was adding some checks and was hoping to keep them separate and keep updating the library itself..

This testcase is essentially the slang/tools/slang-tidy repackaged into slang_package_example repo you had. So, pretty much all the sources should be familiar to you -:)

I have created a small testcase that reproduces it, and attached it if you can take a quick look.  This is a build on Ubuntu BTW  

**Here is the stack trace:**
rogram received signal SIGSEGV, Segmentation fault.
0x000055555558912a in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_data (this=0x0) at /usr/include/c++/11/bits/basic_string.h:195
195	      { return _M_dataplus._M_p; }
(gdb) where
#0  0x000055555558912a in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_data (this=0x0) at /usr/include/c++/11/bits/basic_string.h:195
#1  0x000055555558924f in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_is_local (this=0x0) at /usr/include/c++/11/bits/basic_string.h:230
#2  0x00005555555891a8 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_dispose (this=0x0) at /usr/include/c++/11/bits/basic_string.h:239
#3  0x0000555555585bd2 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string (this=0x0, __in_chrg=<optimized out>) at /usr/include/c++/11/bits/basic_string.h:672
#4  0x0000555555594f92 in slang::SourceManager::LineDirectiveInfo::~LineDirectiveInfo() ()
#5  0x0000555555594fb1 in void std::destroy_at<slang::SourceManager::LineDirectiveInfo>(slang::SourceManager::LineDirectiveInfo*) ()
#6  0x0000555555593332 in void std::_Destroy<slang::SourceManager::LineDirectiveInfo>(slang::SourceManager::LineDirectiveInfo*) ()
#7  0x000055555559104a in void std::_Destroy_aux<false>::__destroy<slang::SourceManager::LineDirectiveInfo*>(slang::SourceManager::LineDirectiveInfo*, slang::SourceManager::LineDirectiveInfo*) ()
#8  0x000055555558e357 in void std::_Destroy<slang::SourceManager::LineDirectiveInfo*>(slang::SourceManager::LineDirectiveInfo*, slang::SourceManager::LineDirectiveInfo*) ()
#9  0x000055555558a677 in void std::_Destroy<slang::SourceManager::LineDirectiveInfo*, slang::SourceManager::LineDirectiveInfo>(slang::SourceManager::LineDirectiveInfo*, slang::SourceManager::LineDirectiveInfo*, std::allocator<slang::SourceManager::LineDirectiveInfo>&) ()
#10 0x0000555555586db5 in std::vector<slang::SourceManager::LineDirectiveInfo, std::allocator<slang::SourceManager::LineDirectiveInfo> >::~vector() ()
#11 0x0000555555597ec8 in slang::SourceManager::FileInfo::~FileInfo() ()
#12 0x00005555558e0395 in slang::SourceManager::SourceManager (this=0x7fffffffd6b0) at /home/srivatsa/Workarea/slang/source/text/SourceManager.cpp:26
#13 0x00005555557f0ae4 in slang::driver::Driver::Driver (this=0x7fffffffd5c0) at /home/srivatsa/Workarea/slang/source/driver/Driver.cpp:39
#14 0x000055555557fb5b in main ()

The part of c++/11 bothers me a bit in the stack trace, dont know if that's a issue. but it compiles and links just fine.

**How to reproduce**
I have attached a slang_package_example.tar.gz which contains a build_slang script which builds slang. Just alter the target location to where you have space.

the slang_package_example is as I've described earlier.   it's just the slang-tidy application with the CMakeLists modified with find_package() and I also setup the compiler directives for Boost Single header.   

It builds just fine. 
my example test is :
top.v: 
module top;
endmodule

just running ./slang-tidy top.sv coredumps and I used GDB to get this stack trace.

Any insight is gratefully appreciated.

[/Sri](
[slang_package_example.tar.gz](https://github.com/user-attachments/files/16075266/slang_package_example.tar.gz)

Practical-UVM-Step-By-Step, 2024-07-04:
hi: I also noticed that this issue is not present when using ninja builds on a mac instead of Unix makefiles. both slang install target and slang_example with ninja builds works. CMake target with unix makefiles seems broken for reasons I dont know.

Also 
Specifically, i found the SLANG_BOOST_SINGLE_HEADER define is set when compiling the main slang library. but you have to set it up in the other example for it to match.  I dont know if this was intentional, hence am requesting feedback as I wasnt 100% sure it was just a documentation issue.

MikePopoloski, 2024-07-04:
Since you're installing a debug build of the slang library you need to also build your application in debug mode by setting `-DCMAKE_BUILD_TYPE=Debug` which should fix the crash.

You don't need to manually add compile definitions like SLANG_BOOST_SINGLE_HEADER or include file directories for the slang library (in fact you should not). CMake takes care of propagating dependency information like that when you link to a cmake target like `slang::slang`

automatic-robot, 2024-07-05:
Thank you Mike. I will close this issue,.

closed 2024-07-05 by automatic-robot