MikePopoloski/slang #1063: Any plans to add CDC linting?

chili-chips-ba, 2024-07-16:
With majority of opensource RTL developers being converted software engineers, a CDC validation tool would greatly help improve the quality of IP in opensource domain.

MikePopoloski, 2024-07-21:
It seems like that would be a good use for slang, but it's not something I'm personally planning on working on any time soon. If you'd like to contribute it I'd be happy to accept a PR.

chili-chips-ba, 2024-07-21:
How about putting it on hold, as opposed to straight closure?!
Also see [this](https://github.com/verilator/verilator/issues/5288#issuecomment-2241091793) and [this](https://github.com/YosysHQ/yosys/issues/4493#issuecomment-2238977629).

MikePopoloski, 2024-07-21:
The issue seemed more like a question (that was then answered), but if you meant it more like a feature request then sure, it can remain open if you prefer.

MikePopoloski, 2024-07-21:
If your hope is that someone will pick up this issue for you it would probably be good to be a lot more clear about what exactly you'd want to see for CDC linting, what kinds of things it would check for, how it might work, etc.

ajeethakv, 2024-07-21:
For syntactic level lint checks, I would imagine we could do similar to what my team and I are doing using Pyslang (named PySlint, UVMLint, SVALint etc. there is even a EMULint version that some folks have branched off from PySlint). But many/some of CDC lint rules would need deeper synthesis aspects, SDC constraints, etc. that may be easier to do with Yosys type of framework (Or Icarus/Verilator after synthesis). If you are interested in volunteering rules for CDC, I can explore implementing some of them on top of Pyslang and opensource.

Regards

chili-chips-ba, 2024-07-21:
Any little bit of CDC help will help. Once embarked on this journey, the mileage may surprise us all 😉

ajeethakv, 2024-07-23:
Sure, please list down what you've in mind. You may also look at creating issues via https://github.com/AsFigo/pyslint/issues

chili-chips-ba, 2024-11-23:
Any update on this?

With our [Wireguard-FPGA](https://github.com/chili-chips-ba/wireguard-fpga/issues/7) project ramping up, we'd like to revisit the question of tool to use for CDC linting.

MikePopoloski, 2024-11-23:
I haven't seen any PRs from you, so no change since the last update. Feel free to be the change you want to see in the world!