MikePopoloski/slang #1071: `hierarchicalPath` does not escaped identifiers

niosHD, 2024-07-22:
I am playing around with slang (via the python bindings) for some time now and I am really impressed with how well it works,  great job! Up until now, slang could deal with everything I threw at it but today I encountered something that I suspect is a bug (or a lack of understanding on my part).

**Describe the bug**
In particular, using pyslang on a netlist I stumbled across a case where the synthesis tool generated module instances (and nets) with identifiers that require escaping. I could parse the respective netlist without any issues but the hierarchical paths that I printed from the script (basically the `hierarchicalPath`) are not correctly escaped.

**To Reproduce**
The following pytest function (and output below) demonstrates the problem. My expectation here, encoded in the last assert, is that I get a valid path from `hierarchicalPath` that I could also parse again (e.g., via `Compilation::tryParseName`) without loosing information.

```python
import pyslang

def test_escaped_signals():
    sv_code = r"""
    `default_nettype none

    module GATE(output wire Z, input wire A);
    endmodule

    module top(output wire X, input wire A);
      wire \wire:in = A;
      wire \wire:out ;
      GATE \inst:complex (.A(\wire:in ), .Z(\wire:out ));
      assign X = \wire:out ;
    endmodule
    """
    tree = pyslang.SyntaxTree.fromText(sv_code, path='test.sv')
    compilation = pyslang.Compilation()
    compilation.addSyntaxTree(tree)

    # Check that the example code is without errors.
    assert len(tree.diagnostics) == 0
    assert len(compilation.getAllDiagnostics()) == 0
    assert len(compilation.getRoot().topInstances) == 1
    root_sym = compilation.getRoot().topInstances[0]

    # Iterate over the body to find the instance. Is there a better way?
    for inst in root_sym.body:
        if inst.kind == pyslang.SymbolKind.Instance:
            break
    else:
        assert False

    # Name is the "plain" value as expected.
    assert inst.name == r"inst:complex"

    # Print hierarchical paths that are not properly escaped.
    print(f"{inst}.hierarchicalPath: '{inst.hierarchicalPath}'")
    for sym in inst.body:
        print(f"{sym}.hierarchicalPath: '{sym.hierarchicalPath}'")

    # Fails here. inst.hierarchicalPath is r"top.inst:complex" which is not what I expected.
    assert inst.hierarchicalPath == r"top.\inst:complex "

```

```txt
-------------------------------- Captured stdout call --------------------------------
Symbol(SymbolKind.Instance, "inst:complex").hierarchicalPath: 'top.inst:complex'
Symbol(SymbolKind.Port, "Z").hierarchicalPath: 'top.inst:complex.Z'
Symbol(SymbolKind.Net, "Z").hierarchicalPath: 'top.inst:complex.Z'
Symbol(SymbolKind.Port, "A").hierarchicalPath: 'top.inst:complex.A'
Symbol(SymbolKind.Net, "A").hierarchicalPath: 'top.inst:complex.A'
============================== short test summary info ===============================
FAILED tests/unit/test_demo.py::test_escaped_signals - AssertionError: assert 'top.inst:complex' == 'top.\\inst:complex '
```

**Additional context**
I encountered this using (py)slang 6.0 and Python 3.11.9 in case it is relevant.

If I am wrong here, and this is actually the intended behavior, then it would be great if somebody can point me in the right direction (i.e., how would I get the properly escaped path?). Thank you!

Regards,
Mario

MikePopoloski, 2024-07-28:
Yeah, this is sort of a bug, or rather an under-specified area of the LRM. The getHierarchicalPath implementation exists most importantly to implement the %m display specifier, and it's not clear how that should behave when escaped identifiers are in use. It seems like other tools elide the backslash even if you use it in the code unless it's actually required to correctly round trip the name, which seems like sane behavior and probably what slang should do as well.

MikePopoloski, 2024-07-28:
Fixed in 15c61a0973c3fc0bb7977c67e542440b0c580cc8

closed 2024-07-28 by MikePopoloski