MikePopoloski/slang #1074: Segmentation fault: 11 when running on Mac Version 13.0.1

ArmenLevoni, 2024-07-29:
**Describe the bug**
I followed the installation from:
https://sv-lang.com/building.html 

**To Reproduce**
Running the tool hits error:
slang -h
slang(6385,0x20c802500) malloc: *** error for object 0x52fee100bd0: pointer being freed was not allocated
slang(6385,0x20c802500) malloc: *** set a breakpoint in malloc_error_break to debug
Abort trap: 6

slang -v  tests/regression/all.sv
Segmentation fault: 11

Please suggest how to fix.

MikePopoloski, 2024-07-31:
I can only provide build support for the latest version of macOS, which is covered by a CI build and so can be verified to work well. I can't reproduce issues from older versions, so you'll need to figure out yourself what's different or broken about your build environment or toolchain that is causing the issue. If you determine there is something broken on the slang side that should be fixed I'm happy to accept a PR.

MikePopoloski, 2024-08-07:
Feel free to reopen if you have more information.

closed 2024-08-07 by MikePopoloski