MikePopoloski/slang #1075: Literal turns into scientific notation when dumping to JSON

naiiren, 2024-08-02:
While trying slang with the test cases from Icarus Verilog, I encountered an issue with literals containing hundreds of bits. One of the cases can be found in [https://github.com/steveicarus/iverilog/blob/master/ivtest/ivltests/ca_pow_unsigned.v](url), the literal `200'd12924697071141057419865760813593169586965814232826232910156` would be converted into scientific notation `200'd129246970711410574198657608135931695869...e22` when dumping to JSON, losing its lower bits.
I checked the file `Json.cpp` but could not identify a quick fix to this problem. Could there be any relevant command-line options I overlooked, or is this an actual issue? Any guidance or assistance in resolving this would be appreciated.

furkanusta, 2024-08-07:
Both this issue and the long decimal in #1077 is caused by the default parameters of `ConstantValue::toString` 

```c++
    std::string toString(
        bitwidth_t abbreviateThresholdBits = SVInt::DefaultStringAbbreviationThresholdBits,
        bool exactUnknowns = false, bool useAssignmentPatterns = false) const;
```
`DefaultStringAbbreviationThresholdBits=128` 
If there are unknowns or the variable is bigger than 128 bits it gets abbreviated representation.

You can see the correct bits if you specify these parameters explicitly. 
```c++

#include "slang/numeric/SVInt.h"
#include "fmt/core.h"

using namespace slang;

int main() {

    SVInt a = SVInt::fromString("96'b111111000000111111000000zzzzzzzzzzzzzzzzz0000011111000001111100000111110000011111000001111100000");
    bitwidth_t MAX_BITS = SVInt::MAX_BITS;
    fmt::println("{}", a.toString());
    fmt::println("{}", a.toString(MAX_BITS));
    fmt::println("{}", a.toString(MAX_BITS, true));
    fmt::println("{}", a.toString(LiteralBase::Binary, true, MAX_BITS));

    fmt::println("");

    SVInt b = SVInt::fromString("200'd12924697071141057419865760813593169586965814232826232910156");
    fmt::println("{}", b.toString());
    fmt::println("{}", b.toString(MAX_BITS));
    fmt::println("{}", b.toString(MAX_BITS, true));
    fmt::println("{}", b.toString(LiteralBase::Binary, true, MAX_BITS));
}
```

Then I suppose question would be, should [ASTSerializer](https://github.com/MikePopoloski/slang/blob/6969c74f11d52f1b58d3eed09890ec70abbff29c/source/ast/ASTSerializer.cpp#L98) print exact bits?

MikePopoloski, 2024-08-08:
Yes, serialized representation should be round-trippable. Fixed in 16501082912762af12604ac4c10614a7ce6a1e53

closed 2024-08-08 by MikePopoloski