MikePopoloski/slang #1077: Literals containing `0x` are not handled correctly while dumping to JSON

naiiren, 2024-08-02:
While trying slang with the test cases from Icarus Verilog, I encountered another issue with literals containing `x`. One of these cases can be found in [array_select.v](https://github.com/steveicarus/iverilog/blob/676b36e4554f119d0606ade79db2fba6834d3445/ivtest/ivltests/array_select.v), where the literal `8'b0x`(Line 52) is dumped into `8'bx` rather than `8'b0x`. The point is that `8'bx` represents `8'bxxxxxxxx` while `8'b0x` actually represents `8'b0000000x`.

I checked `NumberParser.cpp` and it seems fine. So the problem should be relevant to the procedure that dumps the AST to JSON.

naiiren, 2024-08-07:
There is another similar problem, which can be triggered by [pr3197917.v](https://github.com/steveicarus/iverilog/blob/676b36e4554f119d0606ade79db2fba6834d3445/ivtest/ivltests/pr3197917.v), where the long binary literal is converted into a hexadecimal one, during which procedure some digit is consumed by nearby `x` and `z`.

furkanusta, 2024-08-07:
The issue is in the [writeTo](https://github.com/MikePopoloski/slang/blob/6969c74f11d52f1b58d3eed09890ec70abbff29c/source/numeric/SVInt.cpp#L819) function called by `toString`
It omits leading zeroes when printing (stored data is correct).

I've tried this assuming that if there are still bitsLeft and MSB is x/z we have omitted a `0` that should be there.

This fixes the above test case but I am not sure if there are edge cases that I am missing or my assumptions is correct.

```diff
diff --git a/source/numeric/SVInt.cpp b/source/numeric/SVInt.cpp
index e2376837..25ed388e 100644
--- a/source/numeric/SVInt.cpp
+++ b/source/numeric/SVInt.cpp
@@ -842,6 +842,10 @@ void SVInt::writeTo(SmallVectorBase<char>& buffer, LiteralBase base, bool includ
             bitsLeft -= int(shiftAmount);
             x = tmp != 0;
         }
+        if (bitsLeft && (buffer.back() == 'x' || buffer.back() == 'X'
+                         || buffer.back() == 'Z' || buffer.back() == 'z')) {
+            buffer.push_back('0');
+        }
     }
 
     // no digits means this is zero
```

MikePopoloski, 2024-08-07:
Yes, I had started a local fix that looks pretty similar to what you posted.

MikePopoloski, 2024-08-08:
Fixed in 24ec28aac77f4688e14161acc12f420efb306240

closed 2024-08-08 by MikePopoloski