MikePopoloski/slang #1084: Timing check on "logic" System Verilog port

Blebowski, 2024-08-08:
**Describe the bug**
Hi,
I have the following clock gating cell:
```
`timescale 1ps/1ps

module clk_gate_and #(
    parameter G_FPGA_TYPE = "BUFG"  // bypass clock gate for FPGA target
) (
        input  logic clk_in,
        input  logic clk_en,

        output logic clk_out
    );

    always_comb
    begin
        clk_out = clk_en & clk_in;
    end
  
// synthesis translate_off
`ifndef TS_CMN_BLOCKS_SYNTHESIS

    specify
        specparam setup = 10;
        specparam hold = 10;

        $setup(posedge clk_in, clk_en, setup);
        $hold(posedge clk_in, clk_en, hold);
    endspecify

`endif
// synthesis translate_on

endmodule
```

I get an error:
```
../../../../../../../../tools/tropic/ts-common-blocks/v0.32/rtl/sv/clk_gate_and.sv:54:24: error: specify path source 'clk_in' must be an 'input' or 'inout' net or modport port
        $setup(posedge clk_in, clk_en, setup);
                       ^~~~~~
../../../../../../../../tools/tropic/ts-common-blocks/v0.32/rtl/sv/clk_gate_and.sv:14:22: note: declared here
```

Commercial toolchain that we use do not report this issue, normally elaborate and simulate the code.

Is this a bug, or is there some option I need to turn on/off ?

**To Reproduce**
Simply analyze by slang.

MikePopoloski, 2024-08-08:
Yes, this looks like a bug in slang. It's applying the rules for specify path sources also to the arguments to the timing check functions, which is not correct. The fix should be easy.

MikePopoloski, 2024-08-16:
Fixed in 4ba42f0efe8ca003f16fa8bc28a3fd62df1fd62e