MikePopoloski/slang #1104: [bug] Generic interconnect issues

likeamahoney, 2024-08-29:
I am trying to parse an reduced example from LRM 6.6.8 section "Generic interconnect" using `slang` cli. Similar test is presence in `slang` test library - "Configs with libraries and interconnect busses example" from (`../tests/unittests/ast/ConfigTests.cpp`).

Suppose i have a 5 files - `driver.sv  driver.svr  lib.map  nets.pkg  top.sv`

`driver.sv`
```verilog
module driver (output wire logic [0:1] out);
endmodule : driver
```

`driver.svr`
```verilog
module driver
   import NetsPkg::*;
   (output realNet [0:1] out);
endmodule : driver
```

`nets.pkg`
```verilog
package NetsPkg;
   nettype real realNet;
endpackage : NetsPkg
```

`top.sv`
```verilog
module top();
    interconnect aBus[0:3][0:1];
    driver driverArray[0:3](aBus);
endmodule : top
```

And `libmap` file - `lib.map`:

```verilog
library logicLib *.sv;
library realLib *.svr;

config cfgReal;
   design logicLib.top;
   default liblist realLib logicLib;
endconfig

config cfgLogic;
   design logicLib.top;
   default liblist logicLib realLib;
endconfig
```

When i am trying to run `slang` using `cfgLogic` as top module:

```shell
slang --top cfgLogic --libmap=lib.map  nets.pkg
```

It's fails with such an error:

```
Top level design units:
    top

top.sv:3:29: error: cannot connect 'unpacked array [4] of unpacked array [2] of untyped' to each port of type 'logic[0:1]' in 'driverArray'
    driver driverArray[0:3](aBus);
           ~~~~~~~~~~~      ^~~~

Build failed: 1 error, 0 warnings
```

When i am trying to with `cfgReal` top module:

```shell
slang --top cfgReal --libmap=lib.map  nets.pkg
```

`slang` fails with that:

```
Top level design units:
    top

driver.svr:3:12: error: 'realNet' is not a type
   (output realNet [0:1] out);
           ^~~~~~~~~~~~~

Build failed: 1 error, 0 warnings
```

I tried different combinations of options and input files - same errors.

Is this a bug or what could I be doing wrong?

MikePopoloski, 2024-08-30:
It looks like there are two separate bugs here. The first is that the logic for unrolling connections across arrays of instances doesn't seem to handle interconnect nets correctly. The second relates to using a user-defined nettype with an additional packed array dimension. It's not actually clear that you can use a packed array dimension with a nettype of `real` type, but maybe? I'll have to look at the LRM more closely.

likeamahoney, 2024-09-12:
> It looks like there are two separate bugs here. The first is that the logic for unrolling connections across arrays of instances doesn't seem to handle interconnect nets correctly. The second [...]

Hello! Do you any progress with fixing this? Or Can I  start to fixing this by myslef?

MikePopoloski, 2024-09-12:
I am not currently working on it myself.

likeamahoney, 2024-09-16:
> I am not currently working on it myself.

After some research I Considered that example which is presented at LRM 2017 ver. is illegal due to some reasons:

1. All tools which I have been tested (`Questa`, `Cadence`, `VCS`) errors on it
2. https://accellera.mantishub.io/view.php?id=6603 - in that issue it was considered that this example from LRM 2017 illegal because using packed dimensions instead of unpacked in nets of type 'real' is illegal due to 7.4.1 section. So `realNet` `out` output should be declared as unpacked and also `logic` `out` should be declared as unpacked to match `interconnect aBus` dimension (that example was reimplemented at LRM 2023 ver.). 

Whatever I consider that example provided by me is buggy and `slang` works fine with it. Can we close that issue? 

correct example would be: 

`driver.sv`
```verilog
module driver (output wire logic out [0:1]);
endmodule : driver
```

`driver.svr`
```verilog
module driver
   import NetsPkg::*;
   (output realNet out [0:1]);
endmodule : driver
```

`nets.pkg`
```verilog
package NetsPkg;
   nettype real realNet;
endpackage : NetsPkg
```

`top.sv`
```verilog
module top();
    interconnect aBus[0:3][0:1];
    driver driverArray[0:3](aBus);
endmodule : top
```

And `libmap` file - `lib.map`:

```verilog
library logicLib *.sv;
library realLib *.svr;

config cfgReal;
   design logicLib.top;
   default liblist realLib logicLib;
endconfig

config cfgLogic;
   design logicLib.top;
   default liblist logicLib realLib;
endconfig
```

`slang` correctly works with that

```shell
slang --top cfglogic --libmap=lib.map  nets.pkg
slang --top cfgReal --libmap=lib.map  nets.pkg
```

MikePopoloski, 2024-11-03:
After reviewing this I will in fact close it out as not needing any fixes.

closed 2024-11-03 by MikePopoloski