MikePopoloski/slang #1109: Protected member access allowed in external-constraints from class

jrudess, 2024-09-04:
The following example analyzes without error, but since 'a' is a protected member in 'C', I believe there should be a protected member access error in the ```randomize() with``` constraints.

```Systemverilog
class C;
    protected int a;
    rand bit b;
endclass

class T;
    function f();
        C c = new();
        c.randomize() with {
            if (a == 3) {
                b == 1'b1;
            }
        };
    endfunction
endclass
```

MikePopoloski, 2024-09-05:
Hmm, the LRM does not say one way or the other. VCS and Questa error, Xcelium allows it. It makes sense to me to disallow it.

jrudess, 2024-09-05:
VCS used to allow it in older versions as well.  It was something they started to enforce sometime after the 2020 versions I think.

MikePopoloski, 2024-10-05:
Fixed in 2ba4087145919a58ba1221577403c134e7b48e58

closed 2024-10-05 by MikePopoloski