MikePopoloski/slang #1110: [bug] Misinterpretation of the LRM section 6.21 "Scope and lifetime"

likeamahoney, 2024-09-05:
There is an example from 6.21 section of illegal module:

```verilog
module top_illegal; // should not compile

   initial begin

      int svar2 = 2; // static/automatic needed to show intent

      for (int i = 0; i < 3; i++) begin

         int loop3 = 0; // illegal statement

         for (int i = 0; i < 3; i++) begin
            loop3++;
            $display(loop3);
         end
      end
   end
endmodule : top_illegal
```

Slang is just warns on it:

```shell
Top level design units:
    top_illegal
source.sv:5:11: warning: initializing a static variable in a procedural context requires an explicit 'static' keyword [-Wexplicit-static]
      int svar2 = 2; // static/automatic needed to show intent
          ^~~~~
source.sv:9:14: warning: initializing a static variable in a procedural context requires an explicit 'static' keyword [-Wexplicit-static]
         int loop3 = 0; // illegal statement
             ^~~~~
Build succeeded: 0 errors, 2 warnings

compiler returned: 0
slang version 6.0.168+eaaf3e48
```

`VCS` and `Cadence` also just warns on it without error.

But `Siemens Questa` errors:

```shell
-- Compiling module top_illegal
** Warning: design.sv(5): (vlog-2244) Variable 'svar2' is implicitly static. You must either explicitly declare it as static or automatic
or remove the initialization in the declaration of variable.
** Error (suppressible): design.sv(9): (vlog-2244) Variable 'loop3' is implicitly static. You must either explicitly declare it as static or automatic
or remove the initialization in the declaration of variable.
```

Is this done on purpose so that `slang` does not give error on `loop3` initialization for compatibility with `VCS` or is it a lack of analysis in `slang`?

So I think that it should be a warning only on `svar2` initialization but at `loop3` init it should be an error.

MikePopoloski, 2024-09-05:
Yes, it is a warning because most tools allow it and so there is a lot of code out there that assumes it's legal. If you want to be strict about it it's easy enough to add `-Werror=explicit-static` to your command line.

likeamahoney, 2024-09-06:
> Yes, it is a warning because most tools allow it and so there is a lot of code out there that assumes it's legal. If you want to be strict about it it's easy enough to add `-Werror=explicit-static` to your command line.

Thak you for reply! `-Werror=explicit-static` flag will error all assignments in example below:

````
module top_illegal; // should not compile

   initial begin

      int svar2 = 2; // static/automatic needed to show intent

      for (int i = 0; i < 3; i++) begin

         int loop3 = 0; // illegal statement

         for (int i = 0; i < 3; i++) begin
            loop3++;
            $display(loop3);
         end
      end
   end
endmodule : top_illegal
````

But due to standart `int svar2 = 2; ` assignment should be a warning only because it's not inside a loop.

MikePopoloski, 2024-09-06:
Why do you think there's a difference between `svar2` and `loop3` here? Where in the LRM does it say that variables declared inside blocks that are inside loops have special behavior with initializers?

likeamahoney, 2024-09-06:
> Why do you think there's a difference between `svar2` and `loop3` here? Where in the LRM does it say that variables declared inside blocks that are inside loops have special behavior with initializers?

Yes, It is not clear from LRM. But I suppose why it is should be more strict for explicit static variables inside loop. Because the user may expect different behavior from the variable he defined inside the loop (at least that it is initialized to 0 each iteration time for example). But for other variables there is no difference in expected and real behaviour.

MikePopoloski, 2024-09-06:
Those lines apply equally to the other variable as well. Both variables are declared inside a statement block, there's no difference between them. Perhaps a block contained in an `initial` procedure indeed only hits initialization once, but that's not true if the procedure is an always block.

Without some specific reason to make this behavior different I don't see why slang should do so.

likeamahoney, 2024-09-06:
> Those lines apply equally to the other variable as well. Both variables are declared inside a statement block, there's no difference between them. Perhaps a block contained in an `initial` procedure [...]

Thanks for reply! I think that issue can be closed.

closed 2024-09-06 by likeamahoney