MikePopoloski/slang #1112: Drop minimum CMake version to 3.17?

akashlevy, 2024-09-06:
**Is your feature request related to a problem? Please describe.**
When trying to build slang on CentOS/RHEL 7, the latest version of `cmake3` available through standard repositories is version 3.17, which is below version 3.20 (the minimum for slang right now). I tested reducing the minimum version of cmake to 3.17 in `CMakeLists.txt` without any problems (as long as `SLANG_USE_MIMALLOC` is set to `OFF`).

**Describe the solution you'd like**
Drop the minimum CMake version to 3.17. I'm happy to submit a PR for this if you'd like.

**Describe alternatives you've considered**
One could also install a newer CMake on CentOS/RHEL 7. This would have to be from source though.

**Additional context**
CentOS/RHEL 7 is still used at a lot of semiconductor companies, so I think it makes some sense to add support for this.

MikePopoloski, 2024-09-06:
3.17 might work for your specific case but it does not in general; IIRC the Windows build needs at least 3.19, for example. If anything I'm planning to increase the minimum soon now that Ubuntu 24.04 is out and becoming mainstream so I can get rid of some junk in the cmake files.

cmake is very easy to install independently if your system version is too old; besides all of their prebuilt binaries you can also `pip install cmake` via the Python package system and get the latest that way too. I'd prefer if you could do that bit of extra work on your side; RHEL 7 is 10 years old now and I don't have the resources to support systems that old out of the box.

akashlevy, 2024-09-06:
Hi Mike, makes sense to me. Thanks for the tip on `pip install cmake`, I will check that out.

closed 2024-09-06 by akashlevy