MikePopoloski/slang #1115: Multi-dimensional return values from functions?

udif, 2024-09-08:
**Describe the bug**
Should slang support multidimensional functional return values? At the moment it doesn't.

**To Reproduce**
Please see https://github.com/verilator/verilator/issues/5382
Should slang support the spec intent or follow it to the letter?

MikePopoloski, 2024-09-09:
I'm not sure what you're asking for here. slang compiles a multidimensional select of the return value from a function call without issue:
```
module m;                                                                                                                                                                                                                                    
  typedef byte o_t[1][1];                                                                                                                                                                                                                    
  function o_t f1(input byte i);                                                                                                                                                                                                             
    return '{'{1}};                                                                                                                                                                                                                          
  endfunction                                                                                                                                                                                                                                
  function o_t f2;                                                                                                                                                                                                                           
    return '{'{1}};                                                                                                                                                                                                                          
  endfunction                                                                                                                                                                                                                                
                                                                                                                                                                                                                                             
  initial $display(f1(0)[0][0]);  
endmodule
```
Using this syntax (without the parentheses) reports an error but I don't believe it's legal:
```
initial $display(f2[0][0]);  
```
The LRM says:
```
To avoid confusion with hierarchical references into function scopes, parentheses shall always be included
when using a function call as an implicit variable, even if no arguments are being supplied.
```
and
```
It shall be illegal to omit the parentheses in a tf_call unless the subroutine is a task, void function, or class method.
```

udif, 2024-09-09:
Yes, you are right.
The original example had a commented-out section with Verilator's errors, which caused me to miss the fact that the slang error was caused by the 2nd call, which isn't legal.
closing this.

closed 2024-09-09 by udif