MikePopoloski/slang #1123: Bad assignment of integral types to assignment patterns

povik, 2024-09-15:
**Describe the bug**
Bad evaluation of a left-hand side assignment pattern when an integral type is being assigned and the elements of the pattern are wider than a single bit.

**To Reproduce**
Take the following function:

```
function [7:0] f();
	automtic logic [3:0] a, b;
	'{ a, b } = 2'h3;
	return {a, b};
endfunction
```

It evaluates to `8'11xxxxxx` with slang's constant evaluation. I would expect it to be evaluated as `8'00010001` or rejected due to bit width mismatch.

That it might be rejected I base on the following text in LRM 2017 (p. 246):

> When an assignment pattern is used as the left-hand side of an assignment-like context, the positional notation shall be required; and each member expression shall have a bit-stream data type that is assignment compatible with and *has the same number of bits* as the data type of the corresponding element on the right- hand side.

povik, 2024-09-15:
Here's another one related to assignment patterns:

```
function [3:0] f();
	logic [0:0] data[1];
	logic [3:0] a;
	'{ a } = data;
	return a;
endfunction
```

`f()` evaluates to `1'bx` not to `4'bxxxx`as expected. I can open a separate issue but I think the underlying cause is the same.

MikePopoloski, 2024-09-24:
I agree this is a bug, and that a strict reading of the LRM would mean it should be rejected with an error. The only commercial tool I found that supports LHS assignment patterns is Questa, and they do allow the assignment to proceed by doing the expected conversion. I also found an Accellera issue where they request that the rule be relaxed to allow conversions. That makes more sense to me, so I'm inclined to support it that way.

MikePopoloski, 2024-09-29:
Fixed in 9e7611cc575b560c771da0da1c9d44174e0e28ea

closed 2024-09-29 by MikePopoloski