MikePopoloski/slang #1127: -Warith-op-mismatch and 'int unsigned' type in for-loops

jrudess, 2024-09-17:
arith-op-mismatch is not a warning that is enabled by default, but I've been playing around with it to see if I can find any real RTL issues.  The warning seems a bit too aggressive to have much value in large designs.  I'm unsure if there is a way to relax this warning in a way that would make it more useful.

Specifically, anytime ```int``` or ```int unsigned``` are used as for-loop variable types, then it is almost certain that any operations involving the for-loop variable will trigger this warning.

One way to suppress the warning triggered by the example below is to change the control-variable type to ```logic[4:0]```.  But this adds an infinite-loop hazard when param_a is 32, since the for-loop increment at 31 rolls over to 0, and the termination condition is always false.  An additional line of code can be added at the end of the for-loop to detect and prevent the hazard, but that's getting a bit excessive I think:  ```if (i == '1) break;```

```
slangtest180.sv:10:31: warning: arithmetic between operands of different types ('logic[4:0]' and 'int unsigned') [-Warith-op-mismatch]
                count = count + i;
                        ~~~~~ ^ ~
```
```systemverilog
module top;
    localparam param_a = 8;
    logic [4:0] count;
    logic [7:0] valid;

    always_comb begin
        count = 0;
        for (int unsigned i = 0; i < param_a; i++) begin
            if (valid[i]) begin
                count = count + i;
            end
        end
    end
endmodule
```

MikePopoloski, 2024-10-10:
I'm not too surprised that this warning isn't useful on bigger codebases but I'm not sure it can really be tweaked enough to be useful without losing what makes it exist in the first place. The more general width-trunc / width-expand warnings are the ones that are tuned to be less noisy. This one exists in case you really want to find all the places where implicit conversions are happening.

For your loop example, there is the potential for overflow with the addition if param_a is ever increased in size, and more generally speaking you'd have to analyze the whole loop body to know if there is the possibility that the loop induction variable is ever modified in a way that might make that addition potentially overflow, so I can see someone wanting to be notified about that conversion. I don't know what the heuristic would actually be for avoiding the warning in this case.

MikePopoloski, 2024-10-23:
Going to close this until there's some update to help make further progress.

closed 2024-10-23 by MikePopoloski