MikePopoloski/slang #1129: `ifndef in marco

ensilica, 2024-09-18:
**Describe the bug**
A `` `ifndef `` in marco define results in the following error
```
source.sv:3:1: error: missing `endif directive
`DEF_IF_UNDEF(MY_MACRO, 1)
^~~~~~~~~~~~~~~~~~~~~~~~~~
source.sv:1:36: note: expanded from macro 'DEF_IF_UNDEF'
`define DEF_IF_UNDEF(macro, value) `ifndef macro `define macro value `endif
                                   ^~~~~~~
warning: no top-level modules found in design [-Wmissing-top]
Build failed: 1 error, 1 warning

compiler returned: 5
slang version 6.0.173+b2cf6aea
```

**To Reproduce**
Was first observed on local setup but can be easy reproduced on [https://sv-lang.com/explore/](https://sv-lang.com/explore/) by copy and pasting the following code
```
`define DEF_IF_UNDEF(macro, value) `ifndef macro `define macro value `endif

`DEF_IF_UNDEF(MY_MACRO, 1)
```

**Additional context**
Interestingly `` `ifdef `` seems fine
i.e.
```
`define DEF_IF_UNDEF(macro, value) `ifdef macro `define macro value `endif

`DEF_IF_UNDEF(MY_MACRO, 1)
```
creates no error

jrudess, 2024-09-18:
Was wondering why I hadn't hit this, and noticed that you can work around the issue by changing the macro to be multi-line:

```systemverilog
`define DEF_IF_UNDEF(macro, value) \
    `ifndef macro \
        `define macro value \
    `endif
```

MikePopoloski, 2024-09-19:
Yeah, I don't think this is necessarily a bug, more like ambiguity in the body of the macro that you wrote. The `endif is considered to be part of the nested `define body and therefore doesn't terminate the ifndef. You don't get the same error for `ifdef because the macro is not already defined and therefore we skip over tokens until we see the `endif. Putting the endif on its own line makes it so that it's not part of the nested `define. 

The exact behavior here is not specified by the LRM. Some commercial tools like VCS seem to have the same behavior as slang. Do you have a tool that handles it differently? How would you specify what the behavior should be exactly?

ensilica, 2024-09-24:
Yeah I think I'm going to chalk this up to undefine behaviour. I know C/C++ doesn't even allow `idef ` directive in marcos. Probably a good coding rule.

Two Notes:
>Do you have a tool that handles it differently? 

Yes Questa and Xcelium handle this okay 

I also found that the below code worked okay and was one of the reasons I filed a bug.  I would a have expected this to fall over as well but I guess could be considered a useful workaround.

```
`define DEF_IF_UNDEF(macro, value) \
    `ifndef macro \
        `define macro value \
    `endif
```