MikePopoloski/slang #1132: Warnings for parameter width expansion/truncation

jcurtiss8086, 2024-09-23:
During development, I noticed that there is a class of RTL bugs that is missed by Slang warnings related to incorrectly sized parameters. My assumption is that the width-expand and width-trunc warnings purposefully do not cover parameters. Similar to the breakout for port-width-trunc and port-width-expand, can new warnings (param-width-trunc, param-width-expand) be added for explicitly sized parameters?

```systemverilog
module A #(
    bit [3:0] P
);

endmodule

module top;
    localparam bit [2:0] P = '0;

    A #(.P(P)) a();
endmodule
```

MikePopoloski, 2024-09-24:
Yes, this is on purpose. The thinking is that Wconstant-conversion will tell you if an implicit conversion changes the actual value of the constant -- if there's no change then no warning is issued. It would not be hard to add a new warning class for this, though it might be quite noisy unless you want it to specifically apply just to parameters set to the value of other parameters (i.e. you want to ignore setting parameters to literals)?

jcurtiss8086, 2024-09-24:
I've only run into RTL bugs when parameters with packed dimensions are involved.   In the above example, the intention was that module A would receive '{4'h4, 4'h3} but because of the size mismatch it ends up receiving '{4'h2, 4'hB} and -Wconstant-conversion doesn't flag an issue since no bit information is lost.

My original thinking was to apply the warning to all parameter assignments where the LHS and RHS have defined dimensions that are different.  if you think that might be too noisy, then limiting it to assignments involving parameters with packed dimensions is a good middle point.

MikePopoloski, 2024-10-31:
This can now be checked with -Wpacked-array-conv

closed 2024-10-31 by MikePopoloski