MikePopoloski/slang #1133: [bug] UninstantiatedDefSymbol def serialization bug

likeamahoney, 2024-09-23:
Consider a following code example:

```verilog
module debug_rom (
  input  logic         clk_i,
  input  logic         req_i,
  input  logic [63:0]  addr_i,
  output logic [63:0]  rdata_o
);

endmodule

module top (
    input logic clk_i,
    input logic req_i,
    input  logic [63:0]  rom_addr,
    output logic [63:0]  rom_rdata
);

    localparam bit P = 0;

    if (P) begin
        debug_rom i_debug_rom (
            .clk_i,
            .req_i,
            .addr_i  ( rom_addr  ),
            .rdata_o ( rom_rdata )
        );
    end

endmodule
```

`slang` debug build fails on it with assertion:

```shell
internal compiler error: Assertion 'conns.size() == names.size()' failed
  in file /slang/source/ast/symbols/InstanceSymbols.cpp, line 1381
  function: void slang::ast::UninstantiatedDefSymbol::serializeTo(slang::ast::ASTSerializer&) const
```

At release mode it also dumps bad mappings for port connection to names. I think `slang` needs a different approach to serializing this type of ast nodes.

povik, 2024-09-28:
Fwiw I've been using this workaround in yosys-slang: https://github.com/povik/slang/commit/bb66e9a05f023f09b8cfad6d827c5d2d269e7c14

It helped with violations of the `conns.size() == names.size()` invariant though it may not help with the specific example given here.

MikePopoloski, 2024-10-05:
Fixed in 5a4da2c599eb6e1597ecf30a7ff6728ee779c326

closed 2024-10-05 by MikePopoloski