MikePopoloski/slang #1138: [bug] Unpacked union comparing using ternary operator

likeamahoney, 2024-10-01:
Hi, All!

I found an interesting example which based on LRM code sample:

```verilog
module test;

   typedef union tagged {
      struct {
         bit [4:0] reg1, reg2, regd;
      } Add;
   } Instr;

   Instr i1, i2;
   reg e;

   initial begin
      i1 = e ? i1 : i2;
   end

endmodule
```

`slang` error on ternary with:

```shell
unpacked.sv:13:14: error: invalid operands to conditional expression ('test.Instr' and 'test.Instr')
      i1 = e ? i1 : i2;
```

I suppose it's why unpacked unions types canno't be equivalent due to [this](https://github.com/MikePopoloski/slang/blob/1fead9a9a9cd1a689c39234cdaafb554bb5627c3/source/ast/expressions/OperatorExpressions.cpp#L1345)

In LRM section 11.4.11 is is said:

> For all other cases, the type of the first expression and second expression shall be equivalent (see
> 6.22.2).

For unpacked unions, section 6.22.2 defines that:

> An anonymous enum, unpacked struct, or unpacked union type is equivalent to itself among data
> objects declared within the same declaration statement and no other data types.
> struct {int A; int B;} AB1, AB2; // AB1, AB2 have equivalent types
> struct {int A; int B;} AB3; // AB3 is not type equivalent to AB1

So it seems to me that in the example provided `i1` and `i2` are equivalent by type, why is there such a strict rejection of unpacked unions in conditional expressions in `slang`?

MikePopoloski, 2024-10-01:
Hmm, this was explicit added in this commit: https://github.com/MikePopoloski/slang/commit/8c981824e0aac1af4b0c807fdd08410ce6656d43

I can't remember why I did that, and the LRM does seem to support this. Also all major tools I tried allow it too. Probably that diff should be reverted and tests added for the corresponding operations (including constant evaluation).

MikePopoloski, 2024-10-05:
Fixed in 903ed2acb95eff18d704aa8d09d637fd06fc2b37

closed 2024-10-05 by MikePopoloski