MikePopoloski/slang #1141: [bug] Packed array range select assertion

likeamahoney, 2024-10-03:
Hi, all! 

I test such verilog example on `gcc-debug-shared` `slang` build:

```verilog
module test;

  reg [15:0] vect;

  initial begin
    vect[1 -: 4] = 8'b1010_1010;
  end
endmodule
```

It fails with assertion:

```
internal compiler error: Assertion 'key.left <= key.right' failed
  in file /projects/svan-slang/source/../include/slang/util/IntervalMap.h, line 877
  function: slang::uint32_t slang::IntervalMapDetails::LeafNode<TKey, TValue, Capacity>::insertFrom(slang::uint32_t, slang::uint32_t, const slang::IntervalMapDetails::interval<TKey>&, const TValue&) [with TKey = long unsigned int; TValue = const slang::ast::ValueDriver*; unsigned int Capacity = 8; slang::uint32_t = unsigned int]
```

But on release build `slang` correctly reports a range selection error:

```shell
o.sv:6:10: error: cannot select range of [1:-2] from 'reg[15:0]' [-Wrange-oob]
    vect[1 -: 4] = 8'b1010_1010;
```

I think a preliminary check is needed before sending the incorrect value range to the value driver.

MikePopoloski, 2024-10-03:
Hmm yeah, that sounds right.

MikePopoloski, 2024-10-05:
Fixed in 6843b589fdd58a3f3d659f0ade6bbf256b2a222c

closed 2024-10-05 by MikePopoloski