MikePopoloski/slang #1151: -Wconversion Flag Does Not Warn on Ternary Expression

hankhsu1996, 2024-10-10:
**Describe the bug**

When using a ternary expression in SystemVerilog code, slang does not issue a width mismatch warning when the `-Wconversion` flag is enabled. This behavior is inconsistent with direct assignments where the warning is correctly triggered.

**To Reproduce**

Run the following code with slang and the `-Wconversion` flag:

```systemverilog
module test;
  initial begin
    static int a = 3;
    static int b = 5;
    static bit cond = 1;
    static longint result = 7;
    result = cond ? a : b;
  end
endmodule
```

When the assignment uses a ternary expression (`result = cond ? a : b;`), there is no warning, despite the width mismatch. However, when assigning directly without the ternary (`result = a;`), slang correctly triggers a warning.

MikePopoloski, 2024-12-24:
Fixed in dd2addd67be95b542dcd23430dd054cf75c50aa1

hankhsu1996, 2025-01-07:
@MikePopoloski 

I would like to reopen this issue because I believe it is not fully resolved. After further testing, I noticed that when one branch is correct and the other branch involves width expansion, no warning is issued. However, if the implicit conversion results in width truncation, slang does trigger a warning. 

Here is an example that demonstrates the issue:

```systemverilog
module test;
  initial begin
    static int a = 3;
    static int b = 5;
    static bit cond = 1;
    static longint result = 7;
    result = cond ? a : longint'(b);
  end
endmodule
```

In this case, I would expect slang to report a width expansion warning, but now it does not. Could you please take a look at this? 
Thank you!

MikePopoloski, 2025-01-25:
Fixed in 4f1a36c4c7cf87e6580e229a155e0386ea3026d1

closed 2025-01-25 by MikePopoloski