MikePopoloski/slang #1154: Variables in Functions which are used in generate blocks are identified as multi-driven

Badhi, 2024-10-11:
Consider the following code block

```systemverilog
function signed [7:0] incr_sum;
        input signed [7:0] a;

        logic signed [7:0] tmp;
        begin
            tmp = 'd0;
            for (int i=0; i < 8; i=i+1) begin
                tmp = tmp + a[i];
            end
            incr_sum = tmp;
        end
endfunction

module test(input signed [7:0][2:0] a);
    generate genvar m;
    logic [7:0][2:0] temp;
        for (m = 0; m < 3; m=m+1) begin : t
            always_comb begin
                temp[m] = {incr_sum(a[m])};
            end
        end
    endgenerate
endmodule

```

For this slang gives the following error

```
Top level design units:
    test
source.sv:19:28: error: variable 'tmp' driven by always_comb procedure cannot be written to by any other process
                temp[m] = {incr_sum(a[m])};
                           ^~~~~~~~~~~~~~
note: from 'test.t[1]' and 'test.t[0]'
source.sv:8:17: note: original assignment here
                tmp = tmp + a[i];
                ^~~
Build failed: 1 error, 0 warnings

compiler returned: 5
slang version 7.0.19+f31d20e5
```
while other commercial compilers do not recognize this as an issue

MikePopoloski, 2024-10-17:
The LRM is pretty clear that this is not allowed:
```
> 9.2.2.2
The variables assigned on the left-hand side of assignments shall not be assigned by any other
process. This includes variables assigned within functions called by the procedure but not those
assigned within tasks called by the procedure.
```
and
```
> 9.2.2.2.2
Variables on the left-hand side of assignments within an always_comb procedure, including
variables assigned within functions called by the procedure, have restrictions on assignments by
other processes (see 9.2.2.2), whereas always @* permits multiple processes to assign to the same
variable.
```
However, as you mention no commercial tools implement this restriction correctly, or have some other interpretation of it, so you can use the slang flag `--allow-multi-driven-locals` for compatibility.

Badhi, 2024-10-17:
Thanks! Should've referred the LRM. Will try the flag you pointed out

closed 2024-10-17 by Badhi