MikePopoloski/slang #1156: can i analysis a file without type checking?

wzl-1997, 2024-10-16:
hello,  i'm developing a language server based on slang. 
I want to use slang to analysis a single file with out type check, such as error: `use of undeclared identifier 'xxx'`. 
I just want to diagnose syntax error, such as `expected ';' `
I has tried the --lint-only option, and it seems not work. 
Dose slang support this feature? If so, how should i do, pass any options to slang? 
Thanks

likeamahoney, 2024-10-16:
Hello, you should use `--parse-only` cli options I suppose.

MikePopoloski, 2024-10-16:
Yes, `--parse-only` will perform basic syntax checking. That does not include name resolution though; you need to do full type checking to also get name resolution errors reported.