MikePopoloski/slang #1159: Can protected functions become public in extended classes?

jcurtiss8086, 2024-10-20:
Found a case where behavior is different among the big three.   VCS and Questa allows virtual functions to change from protected to public in extended classes (both compilation and simulation).   Xcelium complains at compile time at the definition of C2::f().

Searching the LRM for 'protected' I couldn't find a clear statement which says that the protected attribute carries forward from the parent class to the child regardless of whether the child class includes the keyword.  Using the standard behavior of other programming languages, I suspect that the below code should not be allowed.  But maybe maybe an exception for ```--compat vcs``` is warranted since the LRM seems to be vague here (but it's also highly likely I have missed something hidden in the BNF).

```Systemverilog
class C1;
    virtual protected function void f();
    endfunction
endclass

class C2 extends C1;
    virtual function void f();
    endfunction
endclass

module top;
    C2 c;
    initial begin
        c = new();
        c.f();
    end
endmodule
```

MikePopoloski, 2024-10-20:
You're right, slang should report an error for this case. The LRM says:
```
Virtual method overrides in subclasses shall have matching argument types, identical argument names,
identical qualifiers, and identical directions to the prototype. The virtual qualifier is optional in the
derived class method declarations. The return type of a virtual function shall be either:
— a matching type (see 6.22.1)
— or a derived class type
of the return type of the virtual function in the superclass. It is not necessary to have matching default
expressions, but the presence of a default shall match.
```
The key point missed here is "identical qualifiers", which is what `protected` and `local` are.

MikePopoloski, 2024-11-17:
Fixed in 5d55b0a4c385b62abc4003b8af9a54523ed47862

closed 2024-11-17 by MikePopoloski