MikePopoloski/slang #1161: Can slang support ast.json input and convert it back to system verilog/verilog module files?

markho77, 2024-10-21:
**Is your feature request related to a problem? Please describe.**
We are interested in optimizing AST with other graph-based approaches and convert it back to system verilog/verilog module files for physical design. We found out that slang doesn't have command line option to read in AST and output system verilog/verilog module currently.

**Describe the solution you'd like**
It would be great to have this AST to system verilog/verilog module feature for incorporating slang library with other optimization engines.

**Describe alternatives you've considered**
Currently, we are also looking at YOSYS but it is not as stable as slang.

**Additional context**
N/A

MikePopoloski, 2024-10-21:
It is not currently possible to do this, but I agree it would be a nice feature to have. More generally speaking the serialization stuff for the AST probably needs an overhaul; right now it's pretty specifically coded to just write JSON -- ideally we'd use a more full featured serialization framework that supported multiple input and output formats, since JSON is pretty verbose. It's also the case that the serialized form currently does not include literally *all* of the information represented in the C++ AST, so that would need to be added as well.

Long story short, it's a decent amount of work but nothing that requires deep architectural knowledge or changes to slang, so might be a good project for someone to contribute. For reference there is also [this PR](https://github.com/MikePopoloski/slang/pull/1113) that started adding an AST -> Verilog source pass, which is incomplete but could be a useful thing to flesh out as well.

markho77, 2024-10-22:
Thank you for the quick response!