MikePopoloski/slang #1170: Parameter evaluation error in constant function with array access

hankhsu1996, 2024-11-06:
**Describe the bug**

In the following code, the `Width` parameter, which should be set to `8`, is incorrectly computed as `-1` by slang. This discrepancy only occurs under certain conditions involving array access in constant functions. Other tools (like Verilator) correctly calculate `Width` as `8`.

**To Reproduce**

Steps to reproduce the behavior:

1. Compile the following SystemVerilog code:

    ```systemverilog
    module Test;

      typedef struct packed {
        int width;
      } t_config;

      parameter t_config Config[2] = '{
        0: '{ width: 4 },
        1: '{ width: 8 }
      };

      function automatic int get_max_size(t_config conf[2]);
        int _max = -1;
        for (int i = 0; i < 2; i++) begin
          if (conf[i].width > _max) begin
            _max = conf[i].width;
          end
        end
        return _max;
      endfunction

      parameter int Width = get_max_size(Config);

      logic [Width-1:0] data;
      assign data[0 +: Width] = 0;

      initial begin
        $display("Width: %0d", Width);
        $display("Data: %0h", data);
        $finish();
      end

    endmodule
    ```

2. Observe the following compiler error:
    ```
    test.sv:25:20: error: value must be positive
      assign data[0 +: Width] = 0;
                       ^~~~~
    Build failed: 1 error, 0 warnings
    ```

3. The AST dump shows `Width` as `-1`, while the expected value is `8`.

**Additional context**
This bug only seems to happen when:
1. Constant functions are used in parameters.
2. There’s an array in the parameters.
3. The condition `conf[i].width > _max` inside a `for` loop evaluates as false.
4. If I just return `conf[0].width`, there’s no issue.
5. The problem only appears when accessing `conf[i].width` inside the `for` loop on the RHS.

I tried to simplify the example further, but no luck so far. I’ll post here if I find one.

hankhsu1996, 2024-11-06:
I found a minimal reproducible code snippet:

```systemverilog
module Test;

  typedef struct packed {
    int width;
  } t_config;

  parameter t_config Config = '{ width: 4 };

  $info("Config.width %0d > -1 is evaluated to %0d", Config.width, Config.width > -1);

endmodule
```

Result:
```
Top level design units:
    Test

test.sv:9:3: note: $info encountered: Config.width 4 > -1 is evaluated to 0
  $info("Config.width %0d > -1 is evaluated to %0d", Config.width, Config.width > -1);
  ^

Build succeeded: 0 errors, 0 warnings
```

- This only happens when the struct is packed; there's no issue with an unpacked struct or a scalar number.
- It only produces an incorrect result when comparing with a negative number.

MikePopoloski, 2024-11-07:
Fixed in b3864a1630e176f28faa8c49268b0af75bea4e6a

closed 2024-11-07 by MikePopoloski