MikePopoloski/slang #1174: bitwise-op-parentheses and |= asignments

jcurtiss8086, 2024-11-13:
Is it possible to relax bitwise-op-parentheses when using |= assignments?  I don't think order-of-precedence resolution will be ambiguous for most code-developers in this case and requiring the RHS to be fully wrapped in parentheses isn't necessary.  Otherwise, we have the same lint rule implemented with Spyglass which takes forever to run, so having this warning in Slang is a huge productivity boost for us.

```
bitwise_paren.sv:8:23: warning: '&' within '|' [-Wbitwise-op-parentheses]
            c |= a[i] & b[i];
              ~~ ~~~~~^~~~~~
bitwise_paren.sv:8:23: note: place parentheses around the '&' expression to silence this warning
            c |= a[i] & b[i];
                 ~~~~~^~~~~~
```
```systemverilog
module top;
    logic [15:0] a;
    logic [15:0] b;
    logic c;

    always_comb begin
        for (int i = 0; i < 16; i++) begin
            c |= a[i] & b[i];
        end
    end

endmodule
```

MikePopoloski, 2024-11-14:
Ah yeah, these warnings shouldn't apply to the compound assignment operators.

MikePopoloski, 2024-11-17:
Fixed in 46a5100a2ef7550ac08e0b2f9a06640956680964

closed 2024-11-17 by MikePopoloski