MikePopoloski/slang #1178: [bug] [slang] Severity system tasks with empty arg list

likeamahoney, 2024-11-20:
Hi! Due to grammar defined at LRM (section 20.10) severity system tasks calls with braces but without args is invalid syntax:

> severity_system_task ::=
>   $fatal  [( finish_number [, list_of_arguments ] ) ] ;
> | $error [ ( [ list_of_arguments ] ) ] ;
> | $warning [ ( [ list_of_arguments ] ) ] ;
> | $info [ ( [ list_of_arguments ] ) ] ;

But `slang` and other tools which I was tested doen't errors on that. Was this done on purpose or to maintain compatibility with other tools?

```verilog
module test();
    initial begin
        $fatal(); // no warning
    end
endmodule
```

likeamahoney, 2024-11-20:
Or may be `slang` should emit an error on warning on this but with specific passed cmd flag...

likeamahoney, 2024-11-20:
I described it a little incorrectly in the topic; by system "tasks" I meant only `$fatal`.

MikePopoloski, 2024-11-20:
I suspect the grammar is just incorrect. It makes no sense that the argument is optional, but only if you don't provide the parentheses -- no other argument in the language works like that. Is this causing a problem for you in some way? Not sure how you even noticed it.

likeamahoney, 2024-11-20:
> I suspect the grammar is just incorrect. It makes no sense that the argument is optional, but only if you don't provide the parentheses -- no other argument in the language works like that. Is this causing a problem for you in some way? Not sure how you even noticed it.

Thanks for the answer! I don't think it causing any problem, just for interest. I'm just a follower of strict support for the standard.

I can close the issue.

closed 2024-11-20 by likeamahoney