MikePopoloski/slang #1183: Add 'inside' statements to -Wlogical-not-parentheses

jcurtiss8086, 2024-12-02:
Can the -Wlogical-not-parantheses warning be enhanced to also cover 'inside' statements?  Since SV does not have a 'not_inside' type operator, it is really common to have to add a !() around the inside call and there isn't a warning here when developers forget the parentheses. 

```
inside_paren.sv:4:13: warning: logical not is only applied to the left hand side of this comparison [-Wlogical-not-parentheses]
        if (!a == 3) begin
            ^  ~~
inside_paren.sv:4:14: note: add parentheses after the '!' to evaluate the comparison first
        if (!a == 3) begin
             ^~~~~~
inside_paren.sv:4:13: note: add parentheses around left hand side expression to silence this warning
        if (!a == 3) begin
            ^~
```
```Systemverilog
class C;
    function f();
        bit [7:0] a;
        if (!a == 3) begin
        end
        if (!a inside {0, 3, 9}) begin
        end
    endfunction
endclass
```

jcurtiss8086, 2024-12-04:
Looks like there is some partial overlap with Wcase-outside-range which catches the issue due to the size-mismatch.   But that warning would not flag if the inside-vector only had the values 0 and 1.

```
inside_paren.sv:6:27: warning: 'inside' item with 2 bits can never match the 1 bit case expression [-Wcase-outside-range]
        if (!a inside {0, 3, 9}) begin
                          ^
inside_paren.sv:6:30: warning: 'inside' item with 4 bits can never match the 1 bit case expression [-Wcase-outside-range]
        if (!a inside {0, 3, 9}) begin
                             ^
```

MikePopoloski, 2024-12-05:
Yeah, it should be straightforward to add this and seems useful.

MikePopoloski, 2024-12-15:
Done in 7481d7f5c33ad2e9aa1be7ec267c7914dc87fe3f

closed 2024-12-15 by MikePopoloski